DeviceIoControl/CppKeyboardColour #3: Help To get keyboard effects in Gigabyte g5 laptop

Manish8531, 2024-02-12:


DeviceIoControl, 2024-02-12:
I can take a look into it, but I can't guarantee that I will be able to add support for it.

What software can you use to change the colours on your keyboard?

Manish8531, 2024-02-12:
The Celvo Control Center works, but it only provides me with options to change the color of the keyboard, and only in static mode. Additionally, with this Gigabyte G5 MF laptop model, the keyboard changes its color when it boots, but after that, it stops cycling. So, this hardware is clearly capable. Please, bro, at least help me get one RGB effect. It can be anything. Could you provide a simple, automatic keyboard light on and off feature?
I will be satisfied with that. 
And I clearly know that you are clearly capable of doing this .I have faith in you ❤️
PLEASE HELP

DeviceIoControl, 2024-02-12:
> Additionally, with this Gigabyte G5 MF laptop model, the keyboard changes its color when it boots, but after that, it stops cycling.

Is this after a hard reboot or a soft reboot? (Hard-reboot is a full shutdown and then power the system back on & Soft-reboot is using the 'Restart' option in Windows)

> So, this hardware is clearly capable. Please, bro, at least help me get one RGB effect.

How many RGB zones does this keyboard have?

> The Celvo Control Center works, but it only provides me with options to change the color of the keyboard, and only in static mode.

Did your laptop come with the CLEVO Control Center installed or did you install it manually?

Manish8531, 2024-02-12:
In hard reboot keyboard momentarily cycles through colour
But when I restart it doesn't cycles
It's single zone RGB keyboard
And yes control centre was installed

DeviceIoControl, 2024-02-12:
OK, this is sounding promising and might be something I could sort out. 

I'm going to need a you to get a couple of things from this system, so I know what how to program this.

1.) Could you search for a file called CLEVOMOF.dll inside of either C:\Windows\System32 or C:\Windows\SysWOW64? Just let  me know if it exists or not.

2.) Could you also do a search in the registry for "CLEVOMOF.dll" (without the quotation marks) in HKEY_LOCAL_MACHINE? You can right-click on HKEY_LOCAL_MACHINE and click 'Find'. Make sure you untick everything and only keep "Data" ticked (as shown below)

[image]

Manish8531, 2024-02-13:
**Location of SysWOW64 and celvomof.dll file:**
   - SysWOW64 directory was found.
   - The celvomof.dll file was located within the SysWOW64 directory.

**Procedure followed:**
   - Followed the provided instructions to access the SysWOW64 directory.
   - Conducted a search within the SysWOW64 directory for the celvomof.dll file.

**Screenshot provided:**
   - [image]

   - Apology for the delayed response.
   - Commitment to prompt replies in the future.
   - Billion times Thanks for helping me ❤️

DeviceIoControl, 2024-02-13:
Perfect. Was you able to find the registry key though?

Manish8531, 2024-02-13:
No I am not able to find registry

Manish8531, 2024-02-13:
> Perfect. Was you able to find the registry key though?

Is that compulsory

Manish8531, 2024-02-13:
> Perfect. Was you able to find the registry key though?

If you need anything more than.Please let me know

DeviceIoControl, 2024-02-13:
Hmmm. Interesting. Could you tell me if there is anything in this registry key:

HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Services\WimAcpi. 

Is there a registry value (REG_SZ) called "MofImagePath"?

Manish8531, 2024-02-13:
> > Hmmm. Interesting. Could you tell me if there is anything in this registry key:
> > 
> > HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Services\WmiAcpi. 
> > 
> > Is there a registry value (REG_SZ) called "MofImagePath"?
> > 
> > > Is that compulsory?
> > 
> > Yeah, this is needed as WMI uses this to communicate with your laptops Embedded Controller (EC) - The EC controls the colours on the keyboard.
> > 
> 
> Can you please tell my the paths for registry 

FOUND REGISTRY
AND FOLLOWED YOUR ALL STEPS
HERE ARE THE SCREENSHOTS YOU REQUIRED
[image]

[image]

[image]

And yes there is registery val (REG_SZ) I had also attached a screenshot you can see that for Better reference

DeviceIoControl, 2024-02-13:
OK. Last thing to check.

Could you also screenshot the contents Enum key (inside of WmiAcpi)?
Are there REG_SZ values named 0, 1, & 2 inside of this key and what are the values?

Manish8531, 2024-02-14:
> OK. Last thing to check.
> 
> Could you also screenshot the contents of the Enum key (inside of WmiAcpi)?
> Are there REG_SZ values named 0, 1, & 2 inside of this key and what are the values?
> 

Yes there are values inside Enum keys 
And here is the screenshot
[image]

DeviceIoControl, 2024-02-14:
OK. So it looks like your system seems to have everything in place. 

I will get started on this, but fair warning, I only have a limited amount of time to work on this per day because of my job (I can make more progress on weekends). One thing to also note is it does seem like we are in very different time zones (I'm in the UK), so communication might be a bit lacking on my side.

Now, my plan is to create a lightweight service to change the keyboard colours and a command-line app that can communicate to this service as I'm bad at designing user interfaces. I will leave that up to someone else to do if they would like. (I'm thinking of also designing an API for other developers to interface with if they so desire.)

And since I've not touched this repo in almost 6 years, this code is most definitely going to require refactoring before I start adding support for your system.

I will probably add more comments as time goes on and request for more things if needed.

Manish8531, 2024-02-14:
Thank you for sharing your plan! It sounds like you have a solid approach in mind. Communication timing differences and time constraints are understandable.
I am ok with late response but one request if possible let me aware about the progress.

Regarding the user interface, a simple and functional design will work perfectly fine for me . You  can focus more on functionality and efficiency rather than aesthetics. Just let me know if you need anything 

Thanks for everything ❤️

Manish8531, 2024-02-14:
> Thank you for sharing your plan! It sounds like you have a solid approach in mind. Communication timing differences and time constraints are understandable.
> I am ok with late response but one request if possible keep me aware about the progress.
> 
> Regarding the user interface, a simple and functional design will work perfectly fine for me . You  can focus more on functionality and efficiency rather than aesthetics. Just let me know if you need anything 
> 
> Thanks for everything ❤️

Manish8531, 2024-02-17:
> Thank you for sharing your plan! It sounds like you have a solid approach in mind. Communication timing differences and time constraints are understandable.
> I am ok with late response but one [...]

Hope you are doing well .
If you need anything feel free to ask
Thanks for everything ❤️

DeviceIoControl, 2024-02-18:
OK, the refactor is in progress. It's almost complete, fixing some of the old bugs - I'll push the changes up later on in the week.

Next will be converting the code into a Windows Service and adding a communication mechanism to set the keyboard colours.

Manish8531, 2024-02-19:
> OK, the refactor is in progress. It's almost complete, fixing some of the old bugs - I'll push the changes up later on in the week.
> 
> Next will be converting the code into a Windows Service and adding a communication mechanism to set the keyboard colours.

Thanks ❤️

Manish8531, 2024-02-22:
I hope this message finds you well. 

If possible, could you please provide an update on the progress?

Thanks

DeviceIoControl, 2024-02-22:
Hi,

I am currently doing some testing. I am looking into ways to detect what model of system you are running, so the program can determine if a keyboard theme is playable, along with doing some extra refactoring.

I may need to give you a test program at some point to retrieve the machine's "Board ID" and add it to a list, so I will let you know when that is required.

One thing to also mention is, I am unable to find any way of locating inbuilt themes for your machine. For my machine (Clevo P650RS-G - 3-Zone keyboard), the keyboard theme information can be found in CLEVO Control Center with a little bit of reverse-engineering, but I can't find anything for Single Zone RGB keyboards (which explains why CCC doesn't support it). 

The consequences of this means you would have to make something manually and (depending on how you configure your keyboard theme) it could use between ~12% to ~20% of your CPU when you run it. I mention this because you have a Gaming Laptop and this could affect your gaming performance. With my model of laptop, the inbuilt themes don't use any CPU as that theme is programmed directly into my systems Embedded Controller (EC) and doesn't require any OS intervention.

If you have any questions, please voice them - I will try me best to address them.

Manish8531, 2024-02-22:
> Hi,
>
> I am currently doing some testing. I am looking into ways to detect what model of system you are running, so the program can determine if a keyboard theme is playable, along with doing some [...]

Thanks for the detailed update. I'm perfectly okay with the potential CPU usage and eagerly await the test program. 

Best of luck with the testing and refactoring process! 

Regarding the manual configuration for keyboard themes, I'm ok with it. But please guide me on how to create them?

DeviceIoControl, 2024-02-22:
> Regarding the manual configuration for keyboard themes, I'm ok with it. But please guide me on how to create them?

This will be once I've created program (GUI) to create custom themes.

>  I'm perfectly okay with the potential CPU usage and eagerly await the test program.

Perfect, the test program will just be a small command-line based app you will run on your system (as Admin) and it should give you a number. You will just need to report that back to me and I can add that to the application.

DeviceIoControl, 2024-02-22:
Speaking of themes, are there any keyboard themes you would want me to program directly into the application?

You will need to describe what colours you want the keyboard to be in and its behaviour (colour change timings) etc...

Manish8531, 2024-02-22:
> Speaking of themes, are there any keyboard themes you would want me to program directly into the application?
> 
> You will need to describe what colours you want the keyboard to be in and its behaviour (colour change timings) etc...

It would be fantastic if you could incorporate effects like 

"Color cycle," where the keyboard smoothly transitions through a variety of colors, like red blue green yellow or

 "Breathing," - in which a single colour keyboard light pulsate i.e it on and off

It will great if the effects speed can be controlled 

If those are feasible to implement, then it would be greatly appreciated. 

If not, I trust your judgment to choose other comfortable-to-code options 

Thank you for considering these suggestions!

Mateus-Felipe-Ribeiro, 2024-02-28:
Thanks from Brazil, you are awesome!!

Manish8531, 2024-02-28:
> Thanks from Brazil, you are awesome!!

Absolutely right he is really awesome ❤️

Manish8531, 2024-02-28:
> Speaking of themes, are there any keyboard themes you would want me to program directly into the application?
> 
> You will need to describe what colours you want the keyboard to be in and its behaviour (colour change timings) etc...

Hope you are doing great 
Eagerly waiting for the test program 
Thanks

DeviceIoControl, 2024-02-28:
Hey all,

Sorry for the radio silence, I've been busy at work so I haven't had much time to work on this.

Currently, I am trying to make progress with the test program which will get the "Board ID" for the system, but this looks like it is going to be a bit more tricky than I thought. 

So to explain, the CLEVOMOF.dll library that we located in the SysWOW64 folder is supposed to provide the functionality to set the Keyboard colours, Switch ON/OFF the WiFi, handle the hotkey stuff, etc. It also exposes a method to retrieve the "Board ID" of the system. Unfortunately, because the CLEVO Control Center (and it's related APIs - CLEVOMOF.dll) are very poorly written, the function that I needed to get the Board ID **does not work**. 

Now, I've just reverse engineered the CCC and found another way to do this, but it exists in another DLL as part of the CCC software.

@Manish8531 - Could you see if you can find a DLL called "GetProductdll.dll" inside of the folder - "C:\Program Files (x86)\Hotkey"

Thanks in advance.

Manish8531, 2024-02-29:
> Hey all,
>
> Sorry for the radio silence, I've been busy at work so I haven't had much time to work on this.
>
> Currently, I am trying to make progress with the test program which will get the "Board ID" [...]

Yeah I found out that file its present inside the folder

C:\Program Files (x86)\Hotkey"

Here is the screenshot for your reference

[image]

Thanks

Manish8531, 2024-03-06:
Hello! I hope you're doing well. I'm really eager to hear about the progress of the project.
> Thanks

Manish8531, 2024-03-07:
Hello, I hope everything is fine

DeviceIoControl, 2024-03-07:
Hi,

Once again my apologies for the delay, I've had a lot on my schedule lately.

I have a test program that I will need you to run on your system. This just a console program that will just display some text. Could you just screenshot the output so that I can add this to the program.

I have put the test application into this GitHub repo - "Test\Build\GetProduct.exe".

After that I will build a test program that will test keyboard colours.

Manish8531, 2024-03-07:
> Hi,
> 
> Once again my apologies for the delay, I've had a lot on my schedule lately.
> 
> I have a test program that I will need you to run on your system. This is just a console program that will just display some text. Could you just screenshot the output so that I can add this to the program.
> 
> I have put the test application into this GitHub repo - "Test\Build\GetProduct.exe".
> 
> After that I will build a test program that will test the keyboard colours.
> 
> 

One more question can you please explain me where and how I have to run the code

DeviceIoControl, 2024-03-07:
@Manish8531 
You need to open the highlighted folder in the screenshot below, and run the program called GetProduct.exe

[image]

Please screenshot the output and send it to me when you can. Thanks!

Manish8531, 2024-03-07:
> @Manish8531 
> You need to open the highlighted folder in the screenshot below, and run the program called GetProduct.exe
> 
> [image]
> 
> Please screenshot the output and send it to me when you can. Thanks!
> 

Ok I will share you the screenshot ASAP
Thank you

Manish8531, 2024-03-08:
> > Hi,
> > 
> > Once again my apologies for the delay, I've had a lot on my schedule lately.
> > 
> > I have a test program that I will need you to run on your system. This is just a console program that will just display some text. Could you just screenshot the output so that I can add this to the program.
> > 
> > I have put the test application into this GitHub repo - "Test\Build\GetProduct.exe".
> > 
> > After that I will build a test program that will test the keyboard colours.
> > 
> > 
> 
> One more question can you please explain me where and how I can to run the code 
> Thank you 

Here is the required screenshot of my Device ID

[image]

Thank you for everything
Apologies for late reply

Manish8531, 2024-03-17:
Hope you doing well
Eagerly waiting to hear about the updates 
Thanks

Manish8531, 2024-03-19:
> Here is the required screenshot of my Device ID
>
> [image]
>
> Thank you for everything
> Apologies for late reply

Hope you are well ☺️

DeviceIoControl, 2024-03-23:
Hi,

I'm making some progress, but it has been slow as I need to reverse engineer the newer version of the Clevo Control Center to work out the information that it sends to WMI to change the colour of the keyboards for newer models of CLEVO based laptops.

I can try to make a small program now to test what I have found so far, but I can't guarantee it would work though.

Manish8531, 2024-03-23:
> Hi,
>
> I'm making some progress, but it has been slow as I need to reverse engineer the newer version of the Clevo Control Center to work out the information that it sends to WMI to change the colour [...]

That sounds like a challenging but rewarding project! Your dedication is impressive. 

Eagerly waiting for the test program and hope that it works . 
Good luck, and let me know if you need any assistance along the way!

Thanks

Manish8531, 2024-03-23:
> > Hi,
> > 
> > I'm making some progress, but it has been slow as I need to reverse engineer the newer version of the Clevo Control Center to work out the information that it sends to WMI to change the colour of the keyboards for newer models of CLEVO based laptops.
> > 
> > I can try to make a small program now to test what I have found so far, but I can't guarantee it would work though.
> > 
> 
> That sounds like a challenging but  Your dedication is really impressive. 
> 
> Eagerly waiting for the test program and hope that it works . 
> Good luck, and let me know if you need any assistance along the way!
> 
> Thanks

Manish8531, 2024-03-24:
> > Hi,
> > 
> > I'm making some progress, but it has been slow as I need to reverse engineer the newer version of the Clevo Control Center to work out the information that it sends to WMI to change the colour of the keyboards for newer models of CLEVO based laptops.
> > 
> > I can try to make a small program now to test what I have found so far, but I can't guarantee it would work though.
> > 
> 
> That sounds  challenging but Your dedication is really impressive.  Hats off for your dedication 
> 
> Eagerly waiting for the test program and hope that it works . 
> Good luck, and let me know if you need anything 
> 
> Thanks

Manish8531, 2024-04-03:
> Hi,
>
> I'm making some progress, but it has been slow as I need to reverse engineer the newer version of the Clevo Control Center to work out the information that it sends to WMI to change the colour [...]

Hope you are working on it 
I would be great if share some updates
Thank you

DeviceIoControl, 2024-04-06:
Hi Manish,

I have some good news, I have a test program in the "Test\Build" folder that you can download to test. Could to download "TestKBColour.exe" and see if the keyboard colours change?

Manish8531, 2024-04-07:
> Hi Manish,
> 
> I have some good news, I have a test program in the "Test\Build" folder that you can download to test. Could to download "TestKBColour.exe" and see if the keyboard colours change?
> 

Hi 

I've downloaded the "TestKBColour.exe" from the "Test\Build" folder, but unfortunately, it's not working . The keyboard colors didn't change when I ran the program. Could you please look after it further ?

Thanks,
Manish

DeviceIoControl, 2024-04-07:
Hi Manish,

I forgot to mention that the program needs to be run as Administrator. Was the program run as Administrator?

Manish8531, 2024-04-07:
> Hi Manish,
> 
> I forgot to mention that the program needs to be run as Administrator. Was the program run as Administrator?

Yeah I had

Manish8531, 2024-04-07:
> > Hi Manish,
> > 
> > I forgot to mention that the program needs to be run as Administrator. Was the program run as Administrator?
> 
> Yeah I had run the program as administrator but it didn't work 

I hope you will look after it and fix the problem

Thank you for everything

DeviceIoControl, 2024-04-16:
Hi Manish,

It is possible, I just need to time to reverse engineer the Clevo Control Center to understand exactly what is going on. One thing I remembered is that the CCC uses an Event Log to Inter-process communication (therefore, it might be easier to check those events logs as it has the data it sends to the service).

If I remember correctly, it creates 2 event logs, the problem is that I don't remember the location of them. Could you see if you could search for "OutLog.evtx" and "PowerBiosServerLog.evtx" on your system please? 

If you do have it, please could you use the CCC on your system and change the colours about 5 times (to any colour) and copy and attach the event log to your message if possible?

Manish8531, 2024-04-16:
Here are the files you want 
I hope all the necessary files are their
i have provided all the logs that got generated during the process of colour changing 
and one more thing I have faith in you
You take your time 

If you need anything do let me known

Thanks 

Log Name:      OutLog
Source:        PowerBIOSServer_Out
Date:          16-04-2024 21:03:47
Event ID:      0
Task Category: None
Level:         Information
Keywords:      Classic
User:          N/A
Computer:      ShriRam
Description:
Keyboard_Launch

KB Type: 6
Closing
Event Xml:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="PowerBIOSServer_Out" />
    <EventID Qualifiers="0">0</EventID>
    <Version>0</Version>
    <Level>4</Level>
    <Task>0</Task>
    <Opcode>0</Opcode>
    <Keywords>0x80000000000000</Keywords>
    <TimeCreated SystemTime="2024-04-16T15:33:47.8248160Z" />
    <EventRecordID>139038</EventRecordID>
    <Correlation />
    <Execution ProcessID="19756" ThreadID="0" />
    <Channel>OutLog</Channel>
    <Computer>ShriRam</Computer>
    <Security />
  </System>
  <EventData>
    <Data>Keyboard_Launch

KB Type: 6
Closing</Data>
  </EventData>
</Event>

Log Name:      OutLog
Source:        PowerBIOSServer_Out
Date:          16-04-2024 21:03:47
Event ID:      0
Task Category: None
Level:         Information
Keywords:      Classic
User:          N/A
Computer:      ShriRam
Description:
Keyboard_Launch

KB Type: 6
Closing
Event Xml:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="PowerBIOSServer_Out" />
    <EventID Qualifiers="0">0</EventID>
    <Version>0</Version>
[... 176 lines cut ...]
Computer:      ShriRam
Description:
UpdateAllKBUI R=255 G=0 B=127
Event Xml:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="PowerBIOSServer_Out" />
    <EventID Qualifiers="0">0</EventID>
    <Version>0</Version>
    <Level>4</Level>
    <Task>0</Task>
    <Opcode>0</Opcode>
    <Keywords>0x80000000000000</Keywords>
    <TimeCreated SystemTime="2024-04-16T15:30:00.4342260Z" />
    <EventRecordID>138982</EventRecordID>
    <Correlation />
    <Execution ProcessID="19756" ThreadID="0" />
    <Channel>OutLog</Channel>
    <Computer>ShriRam</Computer>
    <Security />
  </System>
  <EventData>
    <Data>UpdateAllKBUI R=255 G=0 B=127</Data>
  </EventData>
</Event>

Manish8531, 2024-04-16:
The XML codes are missing in the above I tried to post but it's not happening
So follow this link you will get them 

Feel free to ask of anything
Thanks
[]()https://textdoc.co/fpkDw6zlrJExXsu7

DeviceIoControl, 2024-04-16:
Hi Manish,

Would it be possible to upload the evtx files themselves?

Manish8531, 2024-04-16:
will you tell me how to do this

DeviceIoControl, 2024-04-16:
Ahh, no problems, if GitHub doesn't support it, then you don't need to worry about it. Thanks for the info, I will take a look at it.

Manish8531, 2024-04-22:
I hope you're doing well. 
I'm excited to hear about any updates. 
Thank you!

DeviceIoControl, 2024-04-27:
Hi Manish,

Are there any other event logs? 
On my system I have to event logs - OutLog.evtx and PowerBiosServerLog.evtx - Can you find any of these?

I'm looking for an event log that contains event entries that have either one of the following strings:
In this format `"Set WMI=(SomeNumber)  SubCMD=(SomeNumber) Data=(SomeNumber)".`
Or in this format: `w_arg103^(SomeNumber)`

If you can find them that would be great.

Manish8531, 2024-04-28:
> Hi Manish,
> 
> Are there any other event logs? 
> On my system I have two event logs - OutLog.evtx and PowerBiosServerLog.evtx - Can you find any of these?
> 
> I'm looking for an event log that contains event entries that have either one of the following strings:
> In this format `"Set WMI=(SomeNumber)  SubCMD=(SomeNumber) Data=(SomeNumber)".`
> Or in this format: `w_arg103^(SomeNumber)`
> 
> If you can find them that would be great. 

Yeah both outlog.evtx and powerbiosServerLog.evtx is their

DeviceIoControl, 2024-04-28:
OK, I'm going to ask you to use a website called "justbeamit.com". If you can upload the event logs to that and put the link here, I will grab them from you.

justbeamit has a 10 or 15 min limit on how long the link lasts, so as soon as you create the link, please send it, so I can grab the files immediately.

[66 more replies]

closed 2024-08-13 by DeviceIoControl