DeviceIoControl/CppKeyboardColour #4: Implement new themes.

DeviceIoControl, 2024-06-20:
Effect Name: Pulsating Blink

Description:

The pulsating blink effect aims to create a dynamic and attention-grabbing display by alternating between bright and dim states. Here are the specifics of how I envision this effect:

Cycle Pattern:
Start with the zone at full brightness for a brief moment (approximately 0.2 seconds), then quickly fade it to a dim state (approximately 10% brightness) over the next 0.3 seconds. After reaching the dim state, maintain it for about 2 seconds before repeating the cycle.

Smooth Transitions:
Ensure that transitions between brightness levels are smooth and abrupt, creating a noticeable blink effect.

Additional RGB Effect Suggestions:

Breathing Effect: A classic breathing effect where the light gently fades in and out in a chosen color.

Rainbow Sweep: An effect where the colors smoothly transition through the entire RGB spectrum in a continuous sweeping motion.

Random Color:

An effect that randomly changes the color of the zone at regular intervals. Users should be able to set the interval between color changes should be fast like 1 second

Themes suggested by @Manish8531

Manish8531, 2024-06-20:
> Effect Name: Pulsating Blink
>
> Description:
>
> The pulsating blink effect aims to create a dynamic and attention-grabbing display by alternating between bright and dim states. Here are the specifics of [...]

Let's go
I am Ready to assist in every possible way.
Just let me know if you need anything from my side

Thanks

DeviceIoControl, 2024-06-21:
Hi @mehrshad315 - Yes we can add colour more colour, but they will have to be added into the program manually as there is no UI for this application. So if you want more colours, please provide a description of exactly how you how want it to look like and I will see if I can implement it.

Manish8531, 2024-06-21:
> Hi @mehrshad315 - Yes we can add more colours, but they will have to be added into the program manually as there is no UI for this application. So if you want more colours, please provide a description of exactly you how want it to look like and I will see if I can implement it.

Hi @DeviceIoControl ,

Thank you for the clarification. I understand that new colors need to be added manually due to the lack of a UI. Could you please add the following colors for me 

1. **Sky Blue** - `rgb(135, 206, 235)`
2. **Forest Green** - `rgb(34, 139, 34)`
3. **Sunset Orange** - `rgb(255, 69, 0)`
4. **Rose Pink** - `rgb(255, 102, 204)`

Let me know if you need any more details . 

Thank you!

mehrshad315, 2024-06-21:
> Hi @mehrshad315 - Yes we can add more colours, but they will have to be added into the program manually as there is no UI for this application. So if you want more colours, please provide a description of exactly you how want it to look like and I will see if I can implement it.

Thanks for your attention, I understand, so I suggest these colors
rgb(255 255, 0) yellow
rgb(255, 0, 30)  red
rgb(60, 0, 255) Purple
rgb(0 , 255, 60) green
rgb(0 ,255, 120) green
rgb(255,255,255) white
rgb (255, 0 , 255) purple
rgb(255, 120,120) Brown

Manish8531, 2024-07-07:
Hi @DeviceIoControl,

I hope this message finds you well.

I wanted to express my gratitude for previously reducing the transition time to 2 seconds.

 It has made a significant difference. I am writing to kindly request if it would be possible to further reduce the transition time to 1 second. I understand that you may have a busy schedule, but if you could accommodate this change, I would always be gratful to you. 

Thanks for everything 

Lot's of Love

DeviceIoControl, 2024-07-13:
@Manish8531 - Sorry for the late reply, I've been busy - A new build with the 1 sec transition is in "Test/Build/KBInsydeShortBreathe.exe"

Manish8531, 2024-07-14:
> @Manish8531 - Sorry for the late reply, I've been busy - A new build with the 1 sec transition is in "Test/Build/KBInsydeShortBreathe.exe"

I hope you're doing well.

I tested the new build with the 1-second transition, but unfortunately, it is not working. As with the previous build, I ran it as an administrator, but it still did not function correctly.

Could you please look into this issue?

Thank you for your assistance.

DeviceIoControl, 2024-07-14:
Hi,

Are you putting the program in the correct directory with the InsydeDHCU.dll file? If you are could you tell me exactly what is happening with the program?

Manish8531, 2024-07-14:
> Hi,
> 
> Are you putting the program in the correct directory with the InsydeDCHU.dll file? If you are, could you tell me exactly what is happening when you launch the program?
> 
> 

Hi,

Thank you for your assistance. I realized I had forgotten to place the InsydeDCHU.dll file in the correct directory. 

After doing so, the program is now functioning properly.

Thanks for everything

Lot's of Love

mehrshad315, 2024-07-18:
> @Manish8531 - Sorry for the late reply, I've been busy - A new build with the 1 sec transition is in "Test/Build/KBInsydeShortBreathe.exe"

Hello, the KBInsydeShortBreathe.exe file works correctly and smoothly. Thank you very much for your efforts, dear DeviceIoControl.

mehrshad315, 2024-08-04:
Hello, I wanted to ask you, dear DeviceIoControl, can you add different colors to the program or guide me from which part of the color codes to change?

DeviceIoControl, 2024-08-10:
Sorry for the silence - A new build will hopefully be available on Tuesday.

mehrshad315, 2024-08-11:
> Sorry for the silence - A new build will hopefully be available on Tuesday.

Thank you very much for your kindness dear DeviceIoControl

DeviceIoControl, 2024-08-13:
@mehrshad315 - A new build is ready - If you could download Test/Build/KBInsydeBreatheNewColours.exe (and please ensure you have the InsydeDCHU.dll in the same folder.)

@Manish8531 - FYI, I will have a build ready for you on Friday - My apologies again for the delays, I've been busy at work and not been able to work on this project.

mehrshad315, 2024-08-13:
> DeviceIoControl

Hi, everything is working fine and the different colors are showing smoothly, thank you for all your hard work in this long time.

Manish8531, 2024-08-14:
> @mehrshad315 - A new build is ready - If you could download Test/Build/KBInsydeBreatheNewColours.exe (and please ensure you have the InsydeDCHU.dll in the same folder.)
>
> @Manish8531 - FYI, I will [...]

Thank you so much @DeviceIoControl for your efforts in preparing the new build. I truly appreciate all the hard work you’ve put into this. If it’s not too much trouble, could I kindly request the addition of a light blue color  in the build KBInsydeBreatheNewColours.exe ? It would be wonderful if that could be included.

Once again, thank you for everything. I’m grateful for your time and dedication.

Lot's of Love

DeviceIoControl, 2024-08-17:
@Manish8531 - Sorry about the slow progress, I was busy with work and didn't have time yesterday to complete it. I have implemented all of the themes you have requested and I have also added a command line to the program to avoid having multiple programs for each theme.

A new build is ready - If you could download Test/Build/KBInsydeColours.exe (and please ensure you have the InsydeDCHU.dll in the same folder.)

The command-line is as follows:
KBInsydeColours.exe theme [themeName]

Supported command-lines:
`KBInsydeColours.exe theme breathe` (Breathing Effect)
`KBInsydeColours.exe theme colourshift` (For 3-Zone keyboards only).
`KBInsydeColours.exe theme newcolourbreathe` (theme for @mehrshad315 - With the added blue colour as well)
`KBInsydeColours.exe theme colourtransform` (Rainbow Sweep)
`KBInsydeColours.exe theme blink` (Pulsating blink)

(You can do this via Command Prompt.)

NOTE: You can delete the older versions now if you would like as this version has all of the themes in one program.

mehrshad315, 2024-08-18:
Hello, thank you for your efforts, I ran KBInsydeColours, but this error appears and KBInsydeTest does not change the color of the keyboard to green
[image]

DeviceIoControl, 2024-08-18:
@mehrshad315 - You need to use the command-line now as specified in my previous comment. Please read my previous comment carefully.

mehrshad315, 2024-08-18:
I'm sorry I didn't understand, does that mean I should enter the codes from cmd and not use KBInsydeColours.exe to run the themes? Because when I run KBInsydeColours.exe, the error that I sent the picture comes and closes quickly

DeviceIoControl, 2024-08-18:
> I'm sorry I didn't understand, does that mean I should enter the codes from cmd and not use KBInsydeColours.exe to run the themes

Yes, you should do it from CMD instead now, and you can choose a theme based on the provided command lines:

> Supported command-lines:
> KBInsydeColours.exe theme breathe (Breathing Effect)
> KBInsydeColours.exe theme colourshift (For 3-Zone keyboards only).
> KBInsydeColours.exe theme newcolourbreathe (theme for [...]

> (You can do this via Command Prompt.)

mehrshad315, 2024-08-18:
According to your advice, I gave the path of the KBInsydeColours.exe file and the theme code to CMD and the themes worked well. I am really grateful for your efforts and kindness, but can I run this line of code as a shortcut or faster every time?

DeviceIoControl, 2024-08-18:
@mehrshad315  - Yes, you can create a Windows shortcut to KBInsydeColours.exe and set the command line like this:
[image]

mehrshad315, 2024-08-18:
Excuse me, I have another question, is it possible to change the color code number of BreatheAnimationNewColours.cpp like the image below and apply the changes so that later I can change the colors myself without taking your time?
[image]

DeviceIoControl, 2024-08-18:
@mehrshad315 - Yes, if you would like to (and if you know C++).
 
It's open-source, so you can do as you wish with the code. So if there are any new themes you want to create or code by yourself, please do.

mehrshad315, 2024-08-18:
Thank you very much for your reply, this error comes to create a shortcut
[image]

DeviceIoControl, 2024-08-18:
@mehrshad315 - Is the KBInsydeColours.exe in the folder you are specifying? (Did it compile?)

I don't want to clutter this GitHub issue with a bunch of problems irrelevant to the topic, so are there any other means of communication that you have? e.g - Discord, etc...

mehrshad315, 2024-08-18:
Thank you for your help. I was able to create a shortcut for one of the themes by using this URL as an example.
C:\Windows\System32\cmd.exe /k start C:\Users\mehrshad\Desktop\CppKeyboardColour-master\Test\Build\KBInsydeColours.exe theme newcolourbreathe
[image]

mehrshad315, 2024-08-18:
Thank you very much for your patience in answering, wishing you success in all stages of life, with lots of love from Iran.❤️❤️

mehrshad315, 2024-10-24:
@Manish8531 Hello again, after almost 2 months, I wanted to suggest that if you would like to introduce this project on the reddit site, because people who use this model of Gigabyte laptops are still ignorant of this great project, and because I myself am not fluent in English, and if I can't help people's problems

DeviceIoControl, 2024-10-25:
@mehrshad315 - If you would like to post this repo to reddit, then please do. As I mentioned previously, this is open-source so if anyone finds it useful or wants to modify it and create there own version, they are free to do so! 👍

mehrshad315, 2024-10-26:
> @mehrshad315 - If you would like to post this repo to reddit, then please do. As I mentioned previously, this is an open-source project, so if anyone finds it useful or wants to modify it to create their own version, they are free to do so! 👍

Hello, thank you for answering me and allowing us to introduce this project. If I can, I will publish your valuable work on other sites so that everyone can get to know you

Mateus-Felipe-Ribeiro, 2024-11-28:
Hello, can you update the readme with this new commands?

in my laptop don't work, i think my model is old, idk

DeviceIoControl, 2024-11-28:
@Mateus-Felipe-Ribeiro - Have you put the InsydeDCHU.dll into the same folder as the program? What model laptop do you have?

Mateus-Felipe-Ribeiro, 2024-12-10:
You can't belive, in the day when i try, the script don't work. After 2 days restarting the computer, now works hahaha thanks

Manish8531, 2024-12-31:
Dear @DeviceIoControl ,

I hope this message finds you in great spirits. As we bid farewell to 2024, I wanted to take a moment to extend my deepest gratitude to you for your invaluable help in building RGB effects for my laptop keyboard. Despite my keyboard lacking native RGB capabilities, your expertise and dedication transformed my experience, and I cannot thank you enough for the effort and creativity you invested in making it happen.

I also want to sincerely apologize for not reaching out over the past three months. Life kept me quite occupied, but please know that your help has not been forgotten—if anything, it’s cherished even more with each passing day.

Thank you once again for being such a kind and skilled contributor to something that genuinely brought joy to my tech setup. Wishing you a fantastic start to 2025 filled with happiness, success, and more opportunities to share your incredible talents.

Warm regards,
@Manish8531

DeviceIoControl, 2025-01-02:
@Manish8531 - Thanks for the kind words, I really appreciate it. It was a long journey, but we got it working and that's all that matters. 

Hope you've had a great holiday and I hope you also have a great start to 2025.