DeviceIoControl/CppKeyboardColour #6: Help with Clevo Fan Control

LudiusMaximus, 2024-12-19:
Hello, and sorry that I am using the Issues of this project for a more general question.
I am trying to communicate with my Clevo Laptop via event log, just like your programm does, but I seem to be missing something. Any hint would be greatly appreciated.

My Clevo NP70SNC only has a Control Panel software for Windows 11, but I am clinging to Windows 10 for as long as possible. :-)

I found this Clevo Fan Speed app on the Microsoft Store (https://apps.microsoft.com/detail/9p6ljmr12rn1) which can successfully communicate with my laptop hardware.
[image]

Unfortunately, my settings are not persisted over restarts. So I was thinking about a startup script sending the same commands as this Fan Speed app.

In the Event Viewer I see the OutLog events fired whenever the app makes a change.

I tried to emulate these events with a powershell script and in the Event Viewer my events look exactly like the ones triggered by the app, but they have no effect. Do you have any idea what I am missing?

Thanks a lot!

The two events triggered by the app, whenever a change is made: (There are further events when the app starts up, which I have not emulated so far. I thouht emulating these two events should suffice, particularly while the app is still running.)

[image]
[image]

My powershel script:

```powershell
# Define the event details for first event
$eventID1 = 0
$providerName1 = "PowerBIOSServer_Out"
$data1 = "DebugFan:SetWMI Package CMD=14"
$taskCategory1 = 0

# Define the event details for second event
$eventID2 = 0
$providerName2 = "PowerBIOSServer_Out"
$data2 = @"
DebugFan:Set APP Data 
Fan_CPU.T1=51
Fan_CPU.T2=52
Fan_CPU.T3=90
Fan_CPU.T4=100
Fan_CPU.T2_Default=60
Fan_CPU.T3_Default=80
Fan_CPU.D1=23
Fan_CPU.D2=97
Fan_CPU.D3=98
Fan_CPU.D4=100
Fan_CPU.D2_Default=28
Fan_CPU.D3_Default=44
-------------------------------------------------
Fan_GPU1.T1=55
Fan_GPU1.T2=56
Fan_GPU1.T3=78
Fan_GPU1.T4=100
Fan_GPU1.T2_Default=60
Fan_GPU1.T3_Default=80
Fan_GPU1.D1=28
Fan_GPU1.D2=97
Fan_GPU1.D3=98
Fan_GPU1.D4=100
Fan_GPU1.D2_Default=36
 Fan_GPU1.D3_Default=76
-------------------------------------------------
Fan_GPU2.T1=0
Fan_GPU2.T2=100
Fan_GPU2.T3=100
Fan_GPU2.T4=100
Fan_GPU2.T2_Default=0
Fan_GPU2.T3_Default=0
Fan_GPU2.D1=0
Fan_GPU2.D2=100
Fan_GPU2.D3=100
Fan_GPU2.D4=100
Fan_GPU2.D2_Default=1
 Fan_GPU2.D3_Default=2
"@
$taskCategory2 = 0

# Ensure the event log exists
if (-not [System.Diagnostics.EventLog]::SourceExists($providerName1)) {
    New-EventLog -LogName "OutLog" -Source $providerName1
}

# Write the first event log entry
Write-EventLog -LogName "OutLog" -Source $providerName1 -EventID $eventID1 -Category $taskCategory1 -EntryType Information -Message $data1
Write-Output "First event logged successfully."

# Write the second event log entry
Write-EventLog -LogName "OutLog" -Source $providerName2 -EventID $eventID2 -Category $taskCategory2 -EntryType Information -Message $data2
Write-Output "Second event logged successfully."

```

DeviceIoControl, 2024-12-23:
This very much depends on the version of the CCC you have installed. The Inter-process communication method between the CCC and the Hotkey service (which sets the actual keyboard colour) has changed over time.

For the most recent changes I did, my research revealed that the version of the CCC needed to support a Gigabyte single zone laptop used an Invisible Window to communicate and the event log info was not longer being used (although it was still producing events).

LudiusMaximus, 2024-12-23:
Thanks a lot for the reply! Could you give me a hint, how I could find out about how the Clevo App in my case does the communication with the hardware, please? I have a degree in Computer Science, so don't hesitate even if it's complicated. :-)

DeviceIoControl, 2024-12-23:
That would require a bit of reverse engineering (decompilation), so that could take some time.

1.) Install the App (if you haven't already).
2.) Open the folder that the app has been installed to - Windows might say access denied, please let me know if it does, and I will explain how to access it.
3.) Copy the Clevo control center program (*.exe) & the HotKey exe into a different folder.
4.) Download ILSpy
5.) Open the exe files using ILSpy and that should show you the code for the program.

You will need to read the code and work out how it communicates with the hardware.