DeviceIoControl/CppKeyboardColour #7: Problems with installation

RedxzAnomaly, 2024-12-23:
I might be missing something ( maybe im just being dumb), but when I tried testing your program, I encountered an error stating that the Hotkey program DLL was not found. I checked, and it seems that this program isn’t installed on my system. I searched online but couldn’t find a program called "Hotkey." The only similar one I came across was "AutoHotkey."

For context, I’m using a **_G6X-9KG (2024)_** laptop from Gigabyte, and I have both _**G.C.C. (Gigabyte Control Center) and C.C.C. (Clevo Control Center)**_ installed. Could you clarify if the Hotkey program is required and where I might find it? Yes im running as ADM and i checked the Programs (x86) and i dont have the Hotkey installed.

DeviceIoControl, 2024-12-23:
Hi @RedxzAnomaly,

Where did you download the program from?
Could you provide a screenshot of the error please?

RedxzAnomaly, 2024-12-23:
I just installed your repository as a zip file, unzipped it, and that's it. The Clevo Control Center I installed through the following link: `https://www.clevo.com.tw/e-service/download/ftpOut.asp?Lmodel=NHxxDPx&ltype=7`; and the Gigabyte Control Center was pre-installed on my Windows.  
[image]
[image]

DeviceIoControl, 2024-12-23:
Hi @RedxzAnomaly,

You need to run **KBInsydeColours.exe**

The command-line is as follows:
KBInsydeColours.exe theme [themeName]

Supported command-lines:
**KBInsydeColours.exe theme breathe** (Breathing Effect)
**KBInsydeColours.exe theme colourshift** (For 3-Zone keyboards only).
**KBInsydeColours.exe theme newcolourbreathe** (theme for @mehrshad315 - With the added blue colour as well)
**KBInsydeColours.exe theme colourtransform** (Rainbow Sweep)
**KBInsydeColours.exe theme blink** (Pulsating blink)

(You can do this via Command Prompt.)

RedxzAnomaly, 2024-12-23:
Well @DeviceIoControl , i tried to run it in the command line and it didn't work..
[image]

RedxzAnomaly, 2024-12-23:
@DeviceIoControl  I was starting to forget an important lesson I learned from Arch and other Linux distributions: always use the `./` prefix when running commands from a program that isn't directly installed on the operating system. 😂