DeviceIoControl/CppKeyboardColour #9: Quick Question

RedxzAnomaly, 2024-12-24:
Is there any way to run this program in the background? I created a shortcut to execute the command for the breathing effect, but I couldn't find a way to make it run in the background. Any suggestions?

DeviceIoControl, 2024-12-27:
Hi @RedxzAnomaly,

Sorry for the late reply. 

Yes, you can start the program in the background, but you would need to use a program like PSEXEC (SysInternals program) to do so. 

Or if you would like it to launch in the background on system startup, you can register the program as a service using the following command:

`sc create keyboardcolour binPath="PathToKeyboardColourProgram -theme breathe" start=auto type=own`

But there is currently no way to change the active theme while it is running, so you would need to terminate the background process and start the service again using `sc start keyboardcolour`

RedxzAnomaly, 2024-12-27:
@DeviceIoControl Thx bro, im going to use the service option

closed 2024-12-28 by RedxzAnomaly